STMicroelectronics/BlueSTSDK_Python #2: How to get other features or algorithm results using python

titusece, 2018-06-14:
Hello Experts,
I am able to read the temperature, Pressure, Magnetometer, Gyroscope and Accelerometer datas from the STEVAL-WESU1 board using the "example_ble_1.py" python script.
And also I would like to get the algorithm output (like freefall event, Carry Position,  Activity Recognition, accelerometer and pedometer )

Features:
1) Temperature
2) Pressure
3) Magnetometer
4) Gyroscope
5) Accelerometer

Actually I am able to read all the datas using ST android/iOS application, but I would like to get the data using python script.
Can you please help us on this ?

titusece, 2018-06-14:
I have tried the below script to get the notification of USE_CUSTOM_ALGORITHM4 for the freefall event.

```
#define USE_CUSTOM_ALGORITHM4                       1                       //!< define to '1' to enable freefall algorithm usage
#define COPY_ALGORITHM4_BLUEST_CHAR_UUID(uuid_struct)   COPY_UUID_128(uuid_struct,0x00,0x00,0x02,0x00,0x00,0x01,0x11,0xe1,0xac,0x36,0x00,0x02,0xa5,0xd5,0xc5,0x1b)    //!< copy pedometer BlueNRG characteristic UUID

```
And getting the following errors.

I can see this characteristics output from bluetoothctl command.

```
Characteristic
        /org/bluez/hci0/dev_26_80_E1_C6_31_F7/service000c/char0028
        00000200-0001-11e1-ac36-0002a5d5c51b
        Vendor specific
Descriptor
        /org/bluez/hci0/dev_26_80_E1_C6_31_F7/service000c/char0028/desc002a
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration

[WeSU]# select-attribute /org/bluez/hci0/dev_26_80_E1_C6_31_F7/service000c/char0028
[WeSU:/service000c/char0028]# read
Attempting to read /org/bluez/hci0/dev_26_80_E1_C6_31_F7/service000c/char0028
[CHG] Attribute /org/bluez/hci0/dev_26_80_E1_C6_31_F7/service000c/char0028 Value: 0x00
[CHG] Attribute /org/bluez/hci0/dev_26_80_E1_C6_31_F7/service000c/char0028 Value: 0x00
[CHG] Attribute /org/bluez/hci0/dev_26_80_E1_C6_31_F7/service000c/char0028 Value: 0x00
[CHG] Attribute /org/bluez/hci0/dev_26_80_E1_C6_31_F7/service000c/char0028 Value: 0x00
  00 00 00 00                                      ....
[WeSU:/service000c/char0028]#
```

```
Traceback (most recent call last):
  File "st_raspberry_pi_test.py", line 39, in <module>
    svc = p.getServiceByUUID("00000200-0001-11e1-ac36-0002a5d5c51b")
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 449, in getServiceByUUID
    raise BTLEException(BTLEException.GATT_ERROR, "Service %s not found" % (uuid.getCommonName()))
bluepy.btle.BTLEException: Service 00000200-0001-11e1-ac36-0002a5d5c51b not found
```

```
from bluepy import btle
import time

def notify_Write(data):
    global svc, p
    print("Notify write")
    ch_write = svc.getCharacteristics()[0]
    p.writeCharacteristic(ch_write.valHandle, data)
        
class MyDelegate(btle.DefaultDelegate):
    def __init__(self):
        btle.DefaultDelegate.__init__(self)

    def handleNotification(self, cHandle, data):
        print(data)

# Initialisation  -------
p = btle.Peripheral("26:80:E1:C6:31:F7")
#p = btle.Peripheral("26:80:E1:C6:31:F7", "random")
p.setDelegate( MyDelegate() )
# Setup to turn notifications on, e.g.
svc = p.getServiceByUUID("00000400-0001-11e1-ac36-0002a5d5c51b")
ch_notify = svc.getCharacteristics()[1]
notify_Write(b'\x01')

# Main loop --------
while True:
    if p.waitForNotifications(1.0):
        print("Notified!")
        # handleNotification() was called
        continue
    print("Waiting...")
    # Perhaps do something else here
```

titusece, 2018-06-25:
Any response ?

davidroid, 2018-06-25:
Hello Titus,
this if fine with the current status of the Python BlueSTSDK: as you can read in the "Note" Section of the README.md available here https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_Python:

"Currently only a subset of the features is implemented: Switch, Proximity, Accelerometer, Gyroscope, Magnetometer, Pressure, Humidity, Temperature, Second Temperature, Stepper Motor, and Proximity Gesture. Future releases of the Python SDK will cover all the abovementioned features."

Hence all the other features, like those you are looking for, are not available yet; they are in the TODO pipeline and will be added as soon as possible, probably during the next two months.

Please track the webpage for future updates. Thank you.

Regards,
Davide

titusece, 2018-06-26:
Thanks Davide for the response.
Sure, please keep update if any.
Thanks.

closed 2019-10-16 by davidroid