STMicroelectronics/BlueSTSDK_Python #7: Increasing accelerometer  frequency

Gunnybhai, 2019-05-02:
Hello David 
Is it possible to update the sampling frequency of the accelerometer through the python code?

davidroid, 2019-05-02:
Hello,
not, this is not possible via SDK as this feature is not exposed. You can accomplish this goal on either of the following ways:
+ to modify the firmware of the BLE sensors devices;
+ to subsample the notifications coming from the BLE sensors devices (in case you need a lower sampling frequency than the default one);
+ to read (poll) from sensors instead of using notifications from the BLE sensors devices (in case you need a lower sampling frequency than the default one).

Hope this helps. Regards,
Davide

Gunnybhai, 2019-05-02:
Do you know how I could modify the firmware of the BLE sensor device? 
Do you know of any firmware for BLE software updates for over the air?

davidroid, 2019-05-10:
Hello,
I don't know which firmware you are using, but in general you have to find the updating frequency within the code, modify it as per your need, and recompile.

Concerning firmware which supports firmware update, there is this on AI:
https://www.st.com/en/embedded-software/fp-ai-sensing1.html

The BlueSTSDK comes with example number 5 which shows how to perform the firmware update process of precompiled firmware via BLE.

Regards,
Davide

closed 2019-10-08 by davidroid