STMicroelectronics/BlueSTSDK_Python #8: SensorTile not visible and not discoverable.

VitorLima, 2019-06-14:
I am trying to use BlueST-SDK to read and log all sensors from SensorTile device.
There are two main issues:

First: My laptop bluetooth seems to not being able to find the device. I use linux mint OS and it cannot find the SensorTile, using others computers and smartphone it's visible.

Second: When i run the command `sudo python3 Repositories/BlueSTSDK_Python/blue_st_examples/example_ble_1.py`, the result is: 
Scanning Bluetooth devices...
Discovery started.
Bluetooth scanning requires root privilege, so please run the script with "sudo".
Exiting...

but I just used "sudo".
Any idea?

davidroid, 2019-06-14:
Hello Vitor,
Concerning the first issue, can you see other BLE devices when discovering?
Maybe you need to enable the BLE in a configuration panel.

Second, Python3 is not yet supported (it is specified in the README file),
but I am adding it, please keep monitored the package. In the meantime,
please use Python2. Thank you

Regards,
Davide

VitorLima, 2019-06-14:
Thanks for the reply David. It looks like you are right. I don't think my BT hardware can see BLE devices, but I will check it.
The reason I am using Python3 is because I was not able to install pip in Python2. 
I looks like I has and old version but it does not upgrade when I run command.
Any suggestion?

davidroid, 2019-06-14:
Unfortunately I never used Linux mint. I just Googled and found this, like
many other pages, maybe it can be useful:
https://tecadmin.net/install-python-2-7-on-ubuntu-and-linuxmint/

Hope it helps,
Davide

VitorLima, 2019-06-18:
Hi David, looks like if you have Python2 and Python3 at same time, it just install and update pip on Python3. Anyway, fixed that by just copy files. 
Thanks for all the support.
I'd like to ask one more thing. Do think it's possible to log all sensor from SensorTile board at the same time, with this SDK?
Thanks again.

davidroid, 2019-06-18:
Hello Vitor,
The SDK does not provide at the moment a way to log data, but you can
easily implement this by writing manually to a file within you main
application, after polling the sensors or after receiving a notification
from them.

Regards,
Davide

VitorLima, 2019-06-18:
Thank you very much for the kind attention.

closed 2019-06-18 by VitorLima