STMicroelectronics/BlueSTSDK_Python #9: Error running examples in ubuntu

gmaglian, 2019-06-27:
Hello 
few days ago I updated Ubuntu python to version Python 2.7.15+ and can't execute examples.

By running example_ble_1.py I get the following error on library file:
---------------------------------------------------------------------------------------------------------------------
  File "example_ble_1.py", line 51, in <module>
    from blue_st_sdk.manager import Manager
  File "/usr/local/lib/python2.7/dist-packages/blue_st_sdk/manager.py", line 47, in <module>
    from blue_st_sdk.node import Node
  File "/usr/local/lib/python2.7/dist-packages/blue_st_sdk/node.py", line 51, in <module>
    from blue_st_sdk.utils.ble_node_definitions import Debug
  File "/usr/local/lib/python2.7/dist-packages/blue_st_sdk/utils/ble_node_definitions.py", line 41, in <module>
    from blue_st_sdk.features import *
  File "/usr/local/lib/python2.7/dist-packages/blue_st_sdk/features/feature_proximity_gesture.py", line 54, in <module>
    class FeatureProximityGesture(Feature):
  File "/usr/local/lib/python2.7/dist-packages/blue_st_sdk/features/feature_proximity_gesture.py", line 63, in FeatureProximityGesture
    DATA_MAX = len(Gesture) - 1
TypeError: object of type 'type' has no len()
----------------------------------------------------------------------------------------------------------------------------
Pls. can you help?
Thanks
Giovanni Magliano

davidroid, 2019-07-05:
Hello Giovanni,
have you tried re-installing the "enum" package?
sudo pip install enum34

Regards,
Davide

gmaglian, 2019-07-05:
Hello Davide.
I reinstalled the package...  Now it works!

Many thanks for the support.
Ciao
Giovanni

closed 2019-07-09 by gmaglian