STMicroelectronics/BlueSTSDK_Python #10: How to get audio with beamforming ?

EvanFang1013, 2019-09-06:
I can get audio with beamforming on the direction  from ios/android,
but i don't know how to get that function on python.
I have found beamforming of feature on python ,but that only return direction.
I want to get audio data with beamforming, can you teach me how to do ?
please help ,thank you.

matteoronchi, 2019-09-09:
Hi EvanFang1013,

I've just updated the "example_ble_5.py" adding the possibility to activate the beamforming (if it is a feature exported from the BLE node you are using).
I hope that this could solve your issue.

best regards,
Matteo

EvanFang1013, 2019-09-10:
Dear matteoronchi ,
Thanks your solution,
I successfully activate the  beamforming,but i found some bug.
When i activate beamforming then close beamforming ,the sound data become abnormality and LED not flash.And  i have to use IOS APP to recover the Original condition.

Best regards.

closed 2019-09-11 by matteoronchi