STMicroelectronics/BlueSTSDK_Python #11: Predictive Maintenance Feature?

RulerKun, 2019-09-11:
does the BlueSTSDK_Python supports the predictive maintenance?

RulerKun, 2019-09-11:
BlueSTSDK_Android supports it, just wondering if the python version supports it?

davidroid, 2019-09-11:
Hello RulerKun,
The Python version does not support PM yet, but I will align the SDK to the Android version in the next coming weeks. Stay tuned!

Regards,
Davide

RulerKun, 2019-09-11:
Hi @davidroid 

Thanks for the wonderful response!

RulerKun, 2019-10-02:
Hi  
 
How was the progress of development of predictive maintenance for python? is it soon to be released?

davidroid, 2019-10-08:
Hello RulerKun,
the FP-IND-PREDMNT1 is not supported yet, but the predictive maintenance is on the roadmap, so please check here for updates. You can find the supported features in the "Note" section of the "BlueST Protocol" chapter of the readme file.

Thank you and regards,
Davide

RulerKun, 2019-10-28:
Hi

I did some testing on my end. I converted the [FeaturePredictiveSpeedStatus.java](https://github.com/STMicroelectronics/BlueSTSDK_Android/blob/master/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Features/predictive/FeaturePredictiveSpeedStatus.java) and [FeaturePredictiveFrequencyDomainStatus.java](https://github.com/STMicroelectronics/BlueSTSDK_Android/blob/master/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Features/predictive/FeaturePredictiveFrequencyDomainStatus.java) from Java to Python. I also enabled it in the [ble_node_definitions.py](https://github.com/STMicroelectronics/BlueSTSDK_Python/blob/master/blue_st_sdk/utils/ble_node_definitions.py)

Although successfully detected when [example_ble_1.py](https://github.com/STMicroelectronics/BlueSTSDK_Python/blob/master/blue_st_examples/example_ble_1.py) is [Getting features ](https://github.com/STMicroelectronics/BlueSTSDK_Python/blob/ba6fa9514f8d7a39e9e317631137522d906ee44b/blue_st_examples/example_ble_1.py#L235). The data inside the predictive maintenance cannot be fetch.

After some testing i found out that somehow the [BlueSTSDK_Python](https://github.com/STMicroelectronics/BlueSTSDK_Python) cannot enable the extended feature of predictive maintenance. To counter this, i tried enabling the predictive maintenance feature using the [ST BLE Sensor](https://play.google.com/store/apps/details?id=com.st.bluems&hl=en) first and then after enabling it, I run the [example_ble_1.py](https://github.com/STMicroelectronics/BlueSTSDK_Python/blob/master/blue_st_examples/example_ble_1.py), then use the predictive maintenance feature that I converted from Java to Python and successfully managed to get numerical values of predictive maintenance but not status (looks like i didn't managed to successfully convert this)

My question is how can I enable the Feature of predictive maintenance like in this log from Android?

     2019-10-25 09:50:52.449 4178-4178/com.st.BlueSTSDK.Example D/BluetoothGatt: setCharacteristicNotification() - uuid: 00000007-0002-11e1-ac36-0002a5d5c51b enable: true
     2019-10-25 09:51:05.090 4178-4178/com.st.BlueSTSDK.Example D/BluetoothGatt: setCharacteristicNotification() - uuid: 00000007-0002-11e1-ac36-0002a5d5c51b enable: false

This Log was taken from the example application of [BlueSTSDK_Android
](https://github.com/STMicroelectronics/BlueSTSDK_Android)

RulerKun, 2019-11-07:
Hi  
  
Any update on this?

RulerKun, 2019-11-11:
[image]

    from bluepy import btle
    import uuid
    import struct
    class MyDelegate(btle.DefaultDelegate):
        def __init__(self):
            btle.DefaultDelegate.__init__(self)
    
        def handleNotification(self, cHandle, data):
            print("A notification was received: %s" %data)

    p = btle.Peripheral("D9:E5:12:0E:5E:34", "random")
    p.setDelegate( MyDelegate() )

    _NOTIFICATION_ON = struct.pack("BB", 0x01, 0x00)
    _NOTIFICATION_OFF = struct.pack("BB", 0x00, 0x00)
    services=p.getServices()
    while True:
        for service in services:
           print("service ",service)
           characteristics = service.getCharacteristics()
           print("Characteristics ", characteristics)
           for characteristic in characteristics:
               print("characteristic ",characteristic)
               p.writeCharacteristic(characteristic.valHandle+1, _NOTIFICATION_ON)#,True)
               print("on")
               p.waitForNotifications(30.0)
               p.writeCharacteristic(characteristic.valHandle+1, _NOTIFICATION_OFF)#,True)
               print("off")

Is this a correct way in order to enable the predictive maintenance? From the code above. It seems like sometimes the extended features can be enabled, sometimes not

I tried changing [this](https://github.com/STMicroelectronics/BlueSTSDK_Python/blob/ba6fa9514f8d7a39e9e317631137522d906ee44b/blue_st_sdk/node.py#L956) to the example above to like this:

    p.writeCharacteristic(characteristic.valHandle+1, _NOTIFICATION_OFF)

But it seems that its still cannot to enable all of the predictive maintenance features.

Any idea on what parts of the program needed to change in order to enable this predictive maintenace?

closed 2019-11-12 by RulerKun

DiegoManzano1, 2020-01-31:
Hi, how you finally managed predictive maintenance feature?