STMicroelectronics/BlueSTSDK_Python #13: How to choose direction when i activate Beamforming ?

EvanFang1013, 2019-09-19:
Hi~
Excuse me !
I want to choose direction to collect  sound signal,but i don't how to use

By the way,
device.send_command(b"\x00\x00\x08\x00\xAA\x01")
Can you explain the command line about "x00\xAA\x01"
Thank you

matteoronchi, 2019-09-19:
Hi EvanFang1013,

the command structure is the following:
[Feature_Mask | Command_Type | Command_Value]
so, in your case:

[0x00000800 | 0xAA | 0x01]
- Feature_Mask: 0x00000800 --> Beamforming Feature Mask
- Command_Type: 0xAA --> On/Off Beamforming command type
- Command_Value: 0x01 --> Enable

In general, for which concern the Beamforming Feature, the available commands are the following:
- **Enable/Disable Beamforming**:
  - (b"\x00\x00\x08\x00\xAA\x01"), Enable
  - (b"\x00\x00\x08\x00\xAA\x00"), Disable
- **Change Beamforming Direction**:
  - (b"\x00\x00\x08\x00\xBB\x**VV**"), with **VV** in [1,..,8] (8 possible directions if you are using a BlueCoin with the following values:
      - TOP=1
      - TOP_RIGHT=2
      - RIGHT=3
      - BOTTOM_RIGHT=4
      - BOTTOM=5
      - BOTTOM_LEFT=6
      - LEFT=7
      - TOP_LEFT=8
- **Change Beamforming type (Algorithm)**:
  - (b"\x00\x00\x08\x00\xCC\x00"), ASR Ready Beamforming
  - (b"\x00\x00\x08\x00\xCC\x01"), Strong Beamforming

Best regards,
Matteo

EvanFang1013, 2019-09-20:
Dear Matteo,
Thanks for your  support.
I'm successfully.

closed 2019-09-20 by matteoronchi