STMicroelectronics/BlueSTSDK_Python #14: cannot detect the hardware using the program(example_ble_1.py)

RulerKun, 2019-10-08:
Python 3.7.3
Bluetooth 5.0
Raspberry Pi 4B

The Program(example_ble_1.py and example_ble_2.py) cannot detect the sensor (PM1V200) but if using the native bluetooth it manages to detect the sensor(PM1V200)

[image]

davidroid, 2019-10-08:
Dear Rulerkun,
as you can read in the "Setting up the application examples" section of the
repository page, the SDK is compatible only with devices equipped with a
firmware that embeds the BlueST Protocol.

Regards,
Davide

RulerKun, 2019-10-08:
yes, we are using devices equipped with a firmware that embeds the BlueST Protocol.

https://www.st.com/resource/en/product_presentation/fp-ind-predmnt1_quick_start_guide.pdf

 STM32 Nucleo with expansion boards
• NUCLEO-F446RE + X-NUCLEO-CCA02M1 + X-NUCLEO-IDB05A1 + X-NUCLEO-IKS01A2 + STEVAL-MKI182V2

[image]

RulerKun, 2019-10-08:
edit the first comment
- added Raspbian version

davidroid, 2019-10-08:
Ok, now I understand. The FP-IND-PREDMNT1 is not supported yet; you can find the supported features in the "Note" section of the "BlueST Protocol" chapter of the readme file.

We are continuously adding features, and the predictive maintenance is on the roadmap, so please check here for updates.

Thank you for your interest and regards,
Davide

RulerKun, 2019-10-08:
yes although i understand that the preventive maintenance is not yet implemented. We just need to get some basic data like temperature, pressure and humidity for testing purposes. How can we do that? the program can't detect the sensor?

RulerKun, 2019-10-09:
my colleague managed to make it work out with the RPi3, the part where getting the temperature,pressure and humidity but not on the RPi4. Could i know what some parts we need to check so we could fix and make it work in the RPi 4?

davidroid, 2019-10-09:
Don't have at the moment a PI4 to try, but I will manage to try it. There
shouldn't be any difference, anyway.
Are you maybe missing some library?

Regards,
Davide

RulerKun, 2019-10-10:
Hi @davidroid 

I found the problem here, the python 3.7.3 have a problem with the bluepy library. The solution is to downgrade it to python 3.5.7 and now it works.

> Are you maybe missing some library?  

No Sir, i have installed all of the libraries here
- sudo apt-get install python3-pip libglib2.0-dev
- sudo pip3 install bluepy
- sudo pip3 install futures
- sudo pip3 install blue-st-sdk

We are now just waiting for the predictive maintenance feature of BlueSTSDK Python. Can we have the estimated time of release of this feature?

closed 2019-10-16 by RulerKun