STMicroelectronics/BlueSTSDK_Python #17: RuntimeError: can't start new thread

RulerKun, 2019-10-16:
> Traceback (most recent call last):
  File "/home/pi/BlueSTSDK_Python/blue_st_examples/pm3.py", line 329, in <module>
    main(sys.argv[1:])
  File "/home/pi/BlueSTSDK_Python/blue_st_examples/pm3.py", line 273, in main
    device.disable_notifications(feature)
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/node.py", line 986, in disable_notifications
    self.set_notification_status(characteristic, False)
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/node.py", line 942, in set_notification_status
    self._NOTIFICATION_ON if status else self._NOTIFICATION_OFF, True)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 543, in writeCharacteristic
    return self._getResp('wr')
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 416, in _getResp
    self.delegate.handleNotification(hnd, data)
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/node.py", line 1119, in handleNotification
    self._node._update_features(char_handle, data, True)
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/node.py", line 359, in _update_features
    offset += feature.update(timestamp, data, offset, notify_update)
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/feature.py", line 330, in update
    self._notify_update(sample)
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/feature.py", line 274, in _notify_update
    self._thread_pool.submit(listener.on_update(self, sample))
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 160, in submit
    self._adjust_thread_count()
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 181, in _adjust_thread_count
    t.start()
  File "/usr/lib/python3.7/threading.py", line 847, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread
Device PM1V201 disconnected unexpectedly.

> Exiting...

> Exception ignored in: <function Peripheral.__del__ at 0xb6606780>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 630, in __del__
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/node.py", line 559, in disconnect
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/node.py", line 450, in _unexpected_disconnect
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/node.py", line 398, in _update_node_status
  File "/home/pi/BlueSTSDK_Python/blue_st_examples/sample3.py", line 93, in on_disconnect

I was doing some testing where i keep connecting and then disconnecting the program to only one sensor, and somehow i encountered this.  
  
My Program is mostly the same as [example_ble_1.py](https://github.com/STMicroelectronics/BlueSTSDK_Python/blob/master/blue_st_examples/example_ble_1.py) with some less modification with removing the choice part and replacing it with loop of selecting all of them.  
  
I also change the number of notifications to 1, so that the program will only connect to each feature once ( with the exception of the temperature having two temperature)

If my count is right, i have done more or less 70 automated connection and then disconnection.

RulerKun, 2019-10-16:
My goal is to make the program to connect to as many sensors as in the range of bluetooth and loop it back, so the program is not dedicated to connection to only one sensor.

To simulate the connection to as many sensors while having only one sensor, I did make the program connect and then disconnect.

If you have somehow suggestions or recommendation on how to accomplished this. I would appreciate it.

RulerKun, 2019-10-16:
> Traceback (most recent call last):
  File "/home/pi/BlueSTSDK_Python/blue_st_examples/pm3.py", line 329, in <module>
    main(sys.argv[1:])
  File "/home/pi/BlueSTSDK_Python/blue_st_examples/pm3.py", line 290, in main
    if device.wait_for_notifications(0.05):
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/node.py", line 1019, in wait_for_notifications
    return self.waitForNotifications(timeout_s)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 560, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 416, in _getResp
    self.delegate.handleNotification(hnd, data)
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/node.py", line 1119, in handleNotification
    self._node._update_features(char_handle, data, True)
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/node.py", line 359, in _update_features
    offset += feature.update(timestamp, data, offset, notify_update)
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/feature.py", line 330, in update
    self._notify_update(sample)
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/feature.py", line 274, in _notify_update
    self._thread_pool.submit(listener.on_update(self, sample))
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 160, in submit
    self._adjust_thread_count()
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 181, in _adjust_thread_count
    t.start()
  File "/usr/lib/python3.7/threading.py", line 847, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread
Device PM1V201 disconnected.
Exception ignored in: <function Peripheral.__del__ at 0xb65df780>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 630, in __del__
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/node.py", line 555, in disconnect
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/node.py", line 398, in _update_node_status
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 151, in submit
RuntimeError: cannot schedule new futures after shutdown

I got another Error log

davidroid, 2019-10-16:
It seems an error due to the fact that you reached the maximum number of
threads allowed by your system resources.

Davide

RulerKun, 2019-10-16:
try:
        # Creating Bluetooth Manager.
        manager = Manager.instance()
        manager_listener = MyManagerListener()
        manager.add_listener(manager_listener)

        while True:
            # Synchronous discovery of Bluetooth devices.
            print('Scanning Bluetooth devices...\n')
            manager.discover(SCANNING_TIME_s)

            # Getting discovered devices.
            discovered_devices = manager.get_nodes()

            # Listing discovered devices.
            if not discovered_devices:
                print('No Bluetooth devices found. Exiting...\n')
                #sys.exit(0)
            else:
                print('Available Bluetooth devices:')
                i = 1
                for device in discovered_devices:
                    print('%d) %s: [%s]' % (i, device.get_name(), device.get_tag()))
                    i += 1
                
                for device in discovered_devices:
                    node_listener = MyNodeListener()
                    device.add_listener(node_listener)
                    # Connecting to the device.
                    print('Connecting to %s...' % (device.get_name()))
                    if not device.connect():
                        print('Connection failed.\n')
                        continue
                    print(datetimetext)
                    while True:
                        # Getting features.
                        features = device.get_features()
                        print('\nFeatures:')
                        i = 1
                        for feature in features:
                            if isinstance(feature, FeatureAudioADPCM):
                                audio_feature = feature
                                print('%d,%d) %s' % (i,i+1, "Audio & Sync"))
                                i+=1
                            elif isinstance(feature, FeatureAudioADPCMSync):
                                audio_sync_feature = feature
                            else:
                                print('%d) %s' % (i, feature.get_name()))
                                i+=1
                            if feature.get_name() == "Temperature" or feature.get_name() == "Humidity" or feature.get_name() == "Pressure" or feature.get_name() == "Magnetometer" or feature.get_name() == "Gyroscope" or feature.get_name() == "Accelerometer":
                                # Enabling notifications.                    
                                feature_listener = MyFeatureListener()
                                feature.add_listener(feature_listener)
                                device.enable_notifications(feature)
                                # Getting notifications.
                                notifications = 0
                                while notifications < NOTIFICATIONS:
                                    if device.wait_for_notifications(0.05):
                                        notifications += 1
                                # Disabling notifications.
                                device.disable_notifications(feature)
                                feature.remove_listener(feature_listener)                              
                        break                    
                    # Disconnecting from the device.
                    print('\nDisconnecting from %s...' % (device.get_name()))
                    device.remove_listener(node_listener)
                    if not device.disconnect():
                        print('Disconnection failed.\n')
                        continue
            # Resetting discovery.
            manager.reset_discovery()
    except KeyboardInterrupt:
        try:
            # Exiting.
            print('\nExiting...\n')
            sys.exit(0)
        except SystemExit:
            os._exit(0)`

> It seems an error due to the fact that you reached the maximum number of
threads allowed by your system resources.

How will I be able to free some old threads so I could create a new ones?

RulerKun, 2019-10-18:
Hi @davidroid and @mridup   
  
Can i know where part of the program I need to check so I can check whether the Thread previously created was destroyed/Killed when the Feature/Node/Manager was remove from its Listener?

RulerKun, 2019-10-23:
Hi @davidroid 

I did some testing on my end.
> Test 1
I increased the number of NOTIFICATIONS to 1000 and get values from Temperature feature. Getting 1000 temperature values, so far i did not get error

    notifications = 0
    NOTIFICATIONS =100
    while notifications < NOTIFICATIONS:
        if device.wait_for_notifications(0.05):
            notifications += 1

> Test 2
return the number of NOTIFICATIONS to 1
set a 100 loop on getting features [Temperature, Humidity, Pressure, Magnetometer, Gyroscope, Accelerometer] and so far i did not get the RuntimeError: can't start new thread

            num=0
            max =100
            while num< max: 
                for feature in features:
                    if feature.get_name() == "Temperature" or feature.get_name() == "Humidity" or feature.get_name() == "Pressure" or feature.get_name() == "Magnetometer" or feature.get_name() == "Gyroscope" or feature.get_name() == "Accelerometer":
                        # Enabling notifications.                          
                        feature_listener = MyFeatureListener()
                        feature.add_listener(feature_listener)
                        device.enable_notifications(feature)
                        # Getting notifications.
                        notifications = 0
                        while notifications < NOTIFICATIONS:
                            if device.wait_for_notifications(0.05):
                                notifications += 1
                        # Disabling notifications.
                        device.disable_notifications(feature)
                        feature.remove_listener(feature_listener)
                num+=1

> Test 3 
This time i did a 100 loop of device connect, get the features data and disconnect. and did not encounter the runtime error

            num=0
            max =100
            while num < max:
                for device in discovered_devices:
                    node_listener = MyNodeListener()
                    device.add_listener(node_listener)
                    # Connecting to the device.
                    print('Connecting to %s...' % (device.get_name()))
                    if not device.connect():
                        print('Connection failed.\n')
                        continue
                    
                    # Getting features.
                    features = device.get_features()
                    print('Features:')
                    for feature in features:
                        if feature.get_name() == "Temperature" or feature.get_name() == "Humidity" or feature.get_name() == "Pressure" or feature.get_name() == "Magnetometer" or feature.get_name() == "Gyroscope" or feature.get_name() == "Accelerometer":
                            # Enabling notifications.
                            
                            feature_listener = MyFeatureListener()
                            feature.add_listener(feature_listener)
                            device.enable_notifications(feature)
                            # Getting notifications.
                            notifications = 0
                            while notifications < NOTIFICATIONS:
                                if device.wait_for_notifications(0.05):
                                    notifications += 1
                            # Disabling notifications.
                            device.disable_notifications(feature)
                            feature.remove_listener(feature_listener)

                                           
                    # Disconnecting from the device.
                    print('Disconnecting from %s...' % (device.get_name()))
                    device.remove_listener(node_listener)
                    if not device.disconnect():
                        print('Disconnection failed.\n')
                        continue
                num +=1

Could there be a thread in manager class that is not closed properly when rediscovering nodes?