STMicroelectronics/BlueSTSDK_Python #18: example_ble_1.py: Can't find device then timeout and crash

nasrat-v, 2019-10-21:
Hi,

First thanks you for all the great job you've made in this repository.

I run the example_ble_1.py as sudo with Python 3.5.7 to communicate with [STEVAL-BCN002V1 (BlueTile)](https://www.st.com/en/evaluation-tools/steval-bcn002v1b.html)

I'm experiencing following errors :

`##################
# BlueST Example #
##################

Scanning Bluetooth devices...

Discovery started.
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/site-packages/blue_st_sdk/manager.py", line 313, in discover
    self._scanner.scan(timeout_s)
  File "/usr/local/lib/python3.5/site-packages/bluepy/btle.py", line 854, in scan
    self.stop()
  File "/usr/local/lib/python3.5/site-packages/bluepy/btle.py", line 803, in stop
    self._mgmtCmd(self._cmd()+"end")
  File "/usr/local/lib/python3.5/site-packages/bluepy/btle.py", line 312, in _mgmtCmd
    raise BTLEManagementError("Failed to execute management command '%s'" % (cmd), rsp)
bluepy.btle.BTLEManagementError: Failed to execute management command 'scanend' (code: 11, error: Rejected)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./example_ble_1.py", line 312, in <module>
    main(sys.argv[1:])
  File "./example_ble_1.py", line 189, in main
    manager.discover(SCANNING_TIME_s)
  File "/usr/local/lib/python3.5/site-packages/blue_st_sdk/manager.py", line 325, in discover
    raise BlueSTInvalidOperationException(msg)
blue_st_sdk.utils.blue_st_exceptions.BlueSTInvalidOperationException: 
Bluetooth scanning requires root privilege, so please run the script with "sudo".`

It look like errors are from the bluepy packages.
Do you have any idea about how to resolve this issue ?

Thanks.

Regards.

Valentin

RulerKun, 2019-10-21:
have you tried this 

sudo apt-get install python3-pip libglib2.0-dev for python3 or 
sudo apt-get install python-pip libglib2.0-dev for python2

it was part of [Bluepy](https://github.com/IanHarvey/bluepy)

nasrat-v, 2019-10-22:
Hi, thanks for your quick answer.

I'm on Archlinux. 
Glib is installed, same for Python2.7,  3.5 and 3.7. My system is up to date.

If i try it with Python 2.7, I only got blue_st_sdk errors.
I think the problem is coming from archlinux. I probably should install debian or ubuntu ?

~~~~Scanning Bluetooth devices...

Discovery started.
Traceback (most recent call last):
  File "example_ble_1.py", line 312, in <module>
    main(sys.argv[1:])
  File "example_ble_1.py", line 189, in main
    manager.discover(SCANNING_TIME_s)
  File "/usr/lib/python2.7/site-packages/blue_st_sdk/manager.py", line 325, in discover
    raise BlueSTInvalidOperationException(msg)
blue_st_sdk.utils.blue_st_exceptions.BlueSTInvalidOperationException: 
Bluetooth scanning requires root privilege, so please run the script with "sudo".

nasrat-v, 2019-10-22:
On what platform do you use this library ?
I tried Archlinux, Ubuntu and Debian and still the same errors.

Do I need to use a x86 linux ?

RulerKun, 2019-10-23:
I think Debian would be good since i am using Raspbian Buster which is based Debian Buster and the developer of bluepy have develop it in Raspberry Pi (one of the reasons that i choose to use Raspberry Pi) and according to him

> At present it runs on Linux only; I've mostly developed it using a Raspberry Pi, **but it will also run on x86 Debian Linux.**

RulerKun, 2019-10-23:
you could also see this [reply](https://github.com/STMicroelectronics/BlueSTSDK_Python/issues/3#issuecomment-414373795). 

> VM couldn't figure out how to talk to the bluetooth hardware correctly.  

it might be one your configuration on how to talk to the bluetooth hardware

nasrat-v, 2019-10-28:
Okay so I tried on Ubuntu x64, Debian x64 and x86. Still the same error.
I should probably try directly on a Raspbian.

Maybe the problem is coming from the way I install all the depencies (a conflict version between some packets).

musk95, 2019-10-29:
Please use this command 'sudo python3 example_ble_1.py'.

pi@raspberrypi:~ $ python3 example_ble_1.py

##################
# BlueST Example #
##################

Scanning Bluetooth devices...

Discovery started.
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/manager.py", line 313, in discover
    self._scanner.scan(timeout_s)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 852, in scan
    self.start(passive=passive)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 790, in start
    self._mgmtCmd("le on")
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 312, in _mgmtCmd
    raise BTLEManagementError("Failed to execute management command '%s'" % (cmd), rsp)
bluepy.btle.BTLEManagementError: Failed to execute management command 'le on' (code: 20, error: Permission Denied)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "example_ble_1.py", line 312, in <module>
    main(sys.argv[1:])
  File "example_ble_1.py", line 189, in main
    manager.discover(SCANNING_TIME_s)
  File "/usr/local/lib/python3.7/dist-packages/blue_st_sdk/manager.py", line 325, in discover
    raise BlueSTInvalidOperationException(msg)
blue_st_sdk.utils.blue_st_exceptions.BlueSTInvalidOperationException:
Bluetooth scanning requires root privilege, so please run the script with "sudo".

**pi@raspberrypi:~ $ sudo python3 example_ble_1.py**

##################
# BlueST Example #
##################

Scanning Bluetooth devices...

Discovery started.
New device discovered: BCN-002.
Discovery stopped.

Available Bluetooth devices:
1) BCN-002: [e0:f5:f4:fe:de:ad]

nasrat-v, 2019-10-30:
Thanks for you answer.

But as I said in my first comment: "I run the example_ble_1.py as **sudo** with Python 3.5.7". 
I also tried with python2.

RulerKun, 2019-10-31:
could you try adding this line `os.system("sudo service bluetooth start")` [here](https://github.com/STMicroelectronics/BlueSTSDK_Python/blob/ba6fa9514f8d7a39e9e317631137522d906ee44b/blue_st_examples/example_ble_1.py#L179)? and be sure to add import os first

nasrat-v, 2019-12-13:
Ok, I solved the issue.
```systemctl status bluetooth.service``` and ```rfkill list``` showed me that ```rfkill``` was blocking my bluetooth service to work properly.

I just used ```rfkill unblock all``` to unblock the service.
Then the ```exemple_ble.py``` script start working with Python 3.8.

Thanks for help !

closed 2019-12-13 by nasrat-v