STMicroelectronics/BlueSTSDK_Python #20: Device disconnected error in example_ble_1.py

NoahRosa, 2020-03-10:
Hello,

I was trying to run the first example using Python 3.5.2 on a Linux Mint system to communicate with the ST-WB55 and X-NUCLEO-IKS01A3 running the FP-SNS-MOTENVWB1 function pack.

The only modification I did to the code of the example was incrementing the discovery time from 5 to 20 seconds.

Here's the execution:

```
$ sudo python3 example_ble_1.py 

##################
# BlueST Example #
##################

Scanning Bluetooth devices...

Discovery started.
New device discovered: WBMEMS.
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/blue_st_sdk/manager.py", line 313, in discover
    self._scanner.scan(timeout_s)
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 854, in scan
    self.stop()
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 803, in stop
    self._mgmtCmd(self._cmd()+"end")
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 309, in _mgmtCmd
    rsp = self._waitResp('mgmt')
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 362, in _waitResp
    raise BTLEDisconnectError("Device disconnected", resp)
bluepy.btle.BTLEDisconnectError: Device disconnected

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "example_ble_1.py", line 312, in <module>
    main(sys.argv[1:])
  File "example_ble_1.py", line 189, in main
    manager.discover(SCANNING_TIME_s)
  File "/usr/local/lib/python3.5/dist-packages/blue_st_sdk/manager.py", line 325, in discover
    raise BlueSTInvalidOperationException(msg)
blue_st_sdk.utils.blue_st_exceptions.BlueSTInvalidOperationException: 
Bluetooth scanning requires root privilege, so please run the script with "sudo".
```

The same happens also with the second example.

Do you know what causes the device to disconnect?

Thanks,
Noah

fraratto, 2024-12-03:
Hi @NoahRosa, 

I'm facing the same problem (in 2024 sigh), did you find a workaround?

NoahRosa, 2024-12-03:
Hi @fraratto,

Unfortunately since the time I posted this issue I changed company so I no longer have access to what I developed using this product. I remember in the end I made it work with a STM32MP1 as a host device, but I'm not sure if I had to change anything in the library or if it just worked changing the host platform.

I'm sorry for not being more helpful

fraratto, 2024-12-04:
No problem. Thanks for the reply :)