STMicroelectronics/BlueSTSDK_Python #30: Receive data as non-root

aditi1prakash, 2022-03-16:
Current example scripts need root access, is it possible to connect and receive data as non-root?

If not, then why is root access required?

davidroid, 2022-04-14:
Hello Prakash,
On Linux, the Bluetooth protocol stack needs special privileges to interact with. These privileges are implemented through properties called "capabilities" (see [man 7 capability](http://man7.org/linux/man-pages/man7/capabilities.7.html) for details), and the tool to assign capabilities is "setcap".

Since the Python version of the BlueSTSDK makes use of the [bluepy](https://github.com/IanHarvey/bluepy/) library, it is the binary "bluepy-helper" that interacts with the Bluetooth protocol stack; hence, in case you want to do so, you should locate where it is installed and assign the proper capabilities to it. See also [here](https://github.com/IanHarvey/bluepy/issues/313).

Regards,
Davide

aditi1prakash, 2022-04-20:
Hi Davide,

Thanks for the inputs. I modified the capabilities of the bluepy-helper file. 

After assigning the capabilities, the blue_st_sdk throws a **'ModuleNotFoundError'** for Manager module.
[image]

The required capabilities are set for the bluepy-helper file.
[image]

Any thoughts on a fix/workaround would be helpful.

Regards,
Aditi Prakash

davidroid, 2022-04-20:
Hello @aditi1prakash,
Can you check the output of the following command, to see whether the SDK is there?

`find / | grep blue_st_sdk`

Thank you,
Davide