STMicroelectronics/BlueSTSDK_Python #31: Firmware upgrade from different start address in memory

aditi1prakash, 2022-04-20:
Hello, 

The default memory address for firmware upgrade module to upgrade firmware is 0x8004000. 
But there are a few applications from ST that start from different memory location.

Is it possible to configure the SDK to upgrade the firmware starting from a different memory address?

Regards
Aditi Prakash

davidroid, 2022-04-20:
Hello @aditi1prakash,
The BlueSTSDK in fact does not allow to change the memory address because by default the ST firmware knows the location where to start flashing the passed binary, hence it is not needed to configure/change it through the APIs of the SDK.

I will update the description of the "upgrade_firmware()" function of the SDK with this note; thank you for raising a flag here.

Hope this helps. Regards,
Davide

davidroid, 2022-04-21:
> Hello @davidroid <https://github.com/davidroid>,
>
> Thanks for the clarification.
> Do you know how the firmware knows the starting location? Because when I
> use the STM32CubeProgrammer to download the same binary file, I need to
> specify the starting address in the tool.
>
> This led me to believe that starting address is pre-configured in
> BlueSTSDK.
>
> Regards,
> Aditi Prakash

Well,
It's hardcoded in the firmware itself, hence, you have to dig into the
source code of your firmware to find out out.

Regards,
Davide