STMicroelectronics/BlueSTSDK_Python #33: error while excuting example_ble_1

balti09, 2022-07-06:
hi , i'm trying to execute the example_ble_1.py in raspberry pi3 B+ and i get this errors  : 
ps: my python version is 3.9.0
is there any solution ?

`##################
# BlueST Example #
##################

Scanning Bluetooth devices...

Discovery started.
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/blue_st_sdk/manager.py", line 313, in discover
    self._scanner.scan(timeout_s)
  File "/usr/local/lib/python3.9/dist-packages/bluepy/btle.py", line 852, in scan
    self.start(passive=passive)
  File "/usr/local/lib/python3.9/dist-packages/bluepy/btle.py", line 790, in start
    self._mgmtCmd("le on")
  File "/usr/local/lib/python3.9/dist-packages/bluepy/btle.py", line 312, in _mgmtCmd
    raise BTLEManagementError("Failed to execute management command '%s'" % (cmd), rsp)
bluepy.btle.BTLEManagementError: Failed to execute management command 'le on' (code: 20, error: Permission Denied)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/balti/BlueSTSDK_Python/blue_st_examples/example_ble_1.py", line 312, in <module>
    main(sys.argv[1:])
  File "/home/balti/BlueSTSDK_Python/blue_st_examples/example_ble_1.py", line 189, in main
    manager.discover(SCANNING_TIME_s)
  File "/usr/local/lib/python3.9/dist-packages/blue_st_sdk/manager.py", line 325, in discover
    raise BlueSTInvalidOperationException(msg)
blue_st_sdk.utils.blue_st_exceptions.BlueSTInvalidOperationException: 
Bluetooth scanning requires root privilege, so please run the script with "sudo".`

davidroid, 2022-07-06:
Hello,
did you run the script with "sudo"?

Davide

balti09, 2022-07-06:
[image]
yes i run the script with sudo

balti09, 2022-07-06:
> […](#)
[image]
yes i did

balti09, 2022-07-12:
@davidroid is there any solution please ?

davidroid, 2022-07-12:
Hello,
What is the OS you are using on the RPI 3b+? I need to replicate and debug
the issue on the same system.

Regards,
Davide

balti09, 2022-07-13:
Hello @davidroid ,
i'm using the Raspberry Pi OS (https://www.raspberrypi.org/software/) (originally known as Raspbian, a derivative of Debian),
stm board : STEVAL-STWINKT1B
running the example : [FP-IND-PREDMNT1](https://www.st.com/content/st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32-ode-function-pack-sw/fp-ind-predmnt1.html)

hannescam, 2022-12-26:
i have the same problem on archlinux with a pc and my board is a STM32WB55