STMicroelectronics/BlueSTSDK_Python #34: Sto uscendo pazzo -

Martons00, 2024-01-12:
[image]
 
Sto usando un RPi4, devo usare una venv per le librerie ma niente da fare. 
I'm using an RPi4, I need to use a venv for libraries, but I'm not having any luck.

davidroid, 2024-01-12:
Hello Martons00,
This is tricky: when you activate the venv, it automatically sets the PYTHONPATH environment variable to include the venv site-packages directory; since you activate the venv as a normal user but then run the scripts as super user (because the BLE requires this), then the path is not found.

davidroid, 2024-01-12:
Hello Martons00,
This is tricky: when you activate the venv, it automatically sets the PYTHONPATH environment variable to include the venv site-packages directory; since you activate the venv as a normal user but then run the scripts as super user (because the BLE requires this), then the path is not found.
Do become super user before activating the venv, and everything should work:

1. sudo su
2. source .venv/bin/activate
3. python3 example_ble_1.py

Regards,
Davide

Martons00, 2024-01-12:
Thanks you, now it ‘s working
But I can’t find any devices. (SensorTile.BoxPRO)
The sensor works properly with the mobile app(iOS)

davidroid, 2024-01-12:
This is because we are adding the support to this board right now; you can see the list of the currently supported boards here: https://github.com/STMicroelectronics/BlueSTSDK_Python?tab=readme-ov-file#advertising-data
We'll be releasing a new version of the SDK soon, by end of January.

Regards,
Davide

Martons00, 2024-01-12:
[image]

francus-dev, 2024-04-09:
> This is because we are adding the support to this board right now; you can see the list of the currently supported boards here: [...]

Hi Davide, I'm coming through the same issue, and I was becoming crazy too until I read this post and your reply. My enterprise bought the SensorTile.Box Pro to have something more than the standard version, but without the possibility to use the SDK for the acquisition of BTLE data, we are completely stuck to a pole. Is there any information on when will the new SDK be released? 
Regards,
Francesco

francus-dev, 2024-04-18:
> Ah... my bad. Thanks you. I ll wait that version 🍀

Hi Martons00, have you found any solution for this problem? I tried to fix the Python SDK based on information collected in the Android SDK (which already supports the SensorTile.Box Pro), but I don't know why it is not working...

Martons00, 2024-04-18:
I'm waiting for the new version but ... ☠️
I don't know if it will be released