STMicroelectronics/BlueSTSDK_Python #36: Can't connect with SensorTile.box

MilanVercam, 2024-02-27:
Hi,

I want connect my SensorTile box with my Raspberry pi. For this, I use example_ble_1.py, but every time when i try to connect the devices then one of the given errors occur: 

guest@raspberrypi:~ $ sudo python3 example_ble_1.py

##################
# BlueST Example #
##################

Scanning Bluetooth devices...

Discovery started.
New device discovered: TB2    .
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/blue_st_sdk/manager.py", line 313, in discover
    self._scanner.scan(timeout_s)
  File "/usr/local/lib/python3.9/dist-packages/bluepy/btle.py", line 854, in scan
    self.stop()
  File "/usr/local/lib/python3.9/dist-packages/bluepy/btle.py", line 803, in stop
    self._mgmtCmd(self._cmd()+"end")
  File "/usr/local/lib/python3.9/dist-packages/bluepy/btle.py", line 309, in _mgmtCmd
    rsp = self._waitResp('mgmt')
  File "/usr/local/lib/python3.9/dist-packages/bluepy/btle.py", line 362, in _waitResp
    raise BTLEDisconnectError("Device disconnected", resp)
bluepy.btle.BTLEDisconnectError: Device disconnected

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/guest/example_ble_1.py", line 312, in <module>
    main(sys.argv[1:])
  File "/home/guest/example_ble_1.py", line 189, in main
    manager.discover(SCANNING_TIME_s)
  File "/usr/local/lib/python3.9/dist-packages/blue_st_sdk/manager.py", line 325, in discover
    raise BlueSTInvalidOperationException(msg)
blue_st_sdk.utils.blue_st_exceptions.BlueSTInvalidOperationException: 
Bluetooth scanning requires root privilege, so please run the script with "sudo".

guest@raspberrypi:~ $ sudo python3 example_ble_1.py

##################
# BlueST Example #
##################

Scanning Bluetooth devices...

Discovery started.
New device discovered: TB2    .
Discovery stopped.

Available Bluetooth devices:
1) TB2    : [c0:50:06:32:19:33]

Select a device to connect to ('0' to quit): 1
Connecting to TB2    ...
Device TB2     disconnected unexpectedly.

Exiting...

francus-dev, 2025-04-08:
Same identical problem here... and I cannot understand what the cause is: tried both with python3.5 and with python 3.9 and 3.10, but always getting "disconnected unexprectedly".

Martons00, 2025-04-10:
I lost my hopes on this SDK. Same problems.